sd2psXtd/firmware #4: SD2PSX 'freezes' on startup/connect

Slychocobo, 2024-09-14:
Device flashed with 0.9.5-dev (dabe279) but will repeatly 'freeze' on starting up when plugged into a  PS2

Display shows 'Loading card' reaches about 30% then stops updating, does not respond to button presses, is not detected by the PS2.

Removing and reinserting the card will sometimes have to load correctly but this seems to be inconsistant. 

SD card is 16GB Sandisk Ultra Class 10, tried other SD cards with the same result.

Autoboot is Off,
Playstation is a PS2 3900x PAL
Running FMCB  1.966
IDE Drive bay installed with offical PS2 network adaptor.

bbsan2k, 2025-01-29:
Please re-test on RC1

bbsan2k, 2025-03-12:
No repsonse, Issue closed