sd2psXtd/firmware #6: Games with Multiple GameIDs

imdavid555, 2024-12-31:
Having issues with save files for Multi-Disc games.

For example: Star Ocean: Till the End of Time
Disc 1 is SLUS-20488
Disc 2 is SLUS-20891
Unfortunately when booting into Disc 2, it creates a new VMC specific to Disc 2. It does not read the save file from Disc 1.
Was able to work around by renaming the save file.

Are there any plans to support multi-disc games?
Sorry if this is known and was posted else where

AKuHAK, 2025-01-02:
@israpps seems that firmware uses https://github.com/israpps/HDL-Batch-installer/blob/main/Database/gamename.csv that does not contain information about multi-disc games, maybe it will be useful to switch back to the redump database?

dskw, 2025-01-06:
I haven't tried it yet but might have the same issue with Devil May Cry 2, which also has two discs:
Disc 1: SLES-82011
Disc 2: SLES-82012

But we can still navigate between cards manually on the device while the game is booted, so I assume when it comes to the point I could simply switch to the other disc VMC, no?

bbsan2k, 2025-01-25:
Implemented in RC1 - please check Game2Folder mapping feature

closed 2025-01-25 by bbsan2k