sd2psXtd/firmware #7: (1.0.0-beta11) sd2psx stuck on PSX VMC after exiting game

dskw, 2025-01-06:
Release: 1.0.0-beta11 
Device: Bitfunx PSxMemCard Gen2
Title: SLUS-00619 Diablo
Launched through OSD-HDD as a POPS game.

Steps to reproduce:

- Launch game.
- Select "Load Game"
- Notice games are available
- Leave back to main menu without loading
- Notice the card displays the activity border and is stuck
- exit game with IGR

Result:
The card is now stuck on that VMC.
The PS2 browser shows an empty PS1 memory card.
It does not switch to other VMCs when booting a different game.
[image]

Expected behavior:
Existing a POPS game should not lock the mem card to that specific title.

dskw, 2025-01-06:
Not sure if this is the right place for issue reporting for the beta release. Let me know if this needs to go to a different place.

bbsan2k, 2025-01-06:
This is intended behavior; currently there is no way to...
a) detect ps2 from ps1 mode
b) detect a reset of the ps2 console

closed 2025-01-06 by bbsan2k