sd2psXtd/firmware #10: [REPORT]: beta12 breaks ELF autoload from autoboot MC using SwapMagic

rykotxet, 2025-01-15:
### Checks

- [X] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

sd2psx

### PS1 or PS2?

PS2

### [PS2 ONLY]: special consoles

None

### Console model

SCPH-90004

### Firmware version

1.0.0-beta12

### related game

SwapMagic 3.8 Coder (PAL)

### Explain your issue

I use SwapMagic DVD to autoload [PS2BBL](https://github.com/israpps/PlayStation2-Basic-BootLoader) on boot, placed in the autoboot MC of my SD2PSX. It worked fine with the former firmware I had installed ([unofficial 0.9.3](https://github.com/bbsan2k/sd2psx_firmware/releases/tag/unofficial_0.9.3)), but with this beta12 it quickly loads another MC after the autoboot MC (in this case firmware seems to recognize SwapMagic as "Dead or Alive 2 - Hardcore" [SCES-50003]) before SwapMagic can launch PS2BBL from autoboot MC.

The expected behavior would be to keep the autoboot MC loaded long enough to let SwapMagic access that autoboot MC instead of the MC matching its internal serial code.

bbsan2k, 2025-03-12:
Can you please check latest nightly, if this issue persists?

rykotxet, 2025-03-14:
Hi @bbsan2k, I just tested the latest nightly and the issue persists. When the PlayStation 2 logo splashes on screen, the BOOT MC starts to change to the MC of Dead of Alive 2 - Hardcore.

bbsan2k, 2025-05-30:
Okay, I just tried to figure out what's happening here...
Actually it seems, that sd2psx is basically too quick in startup 😃 
SwapMagic seems to utilize the DOA2 Game ID, so it's written to history - and sd2psx is picking up this history write.
You probably have some options:
- Use Game2Folder.ini to map DOA2 Game ID to BOOT folder (although this will send you to BootCard as well if you are really playing DOA2)
- Use PS2BBL through OpenTuna/Funtuna or some other variant that works on 90k 2.3
- Turn Off Game ID

TBH I would go with option 1 or 2 😃

closed 2025-06-04 by bbsan2k

rykotxet, 2025-07-20:
Well, I just wanted to say thanks for the effort, @bbsan2k. You're doing great supporting this device. To be honest, I prefer the game ID in SD2PSX over the SwapMagic's autoload on startup.