sd2psXtd/firmware #11: [ISSUE]: PS1 on PS2 via DKWDRV

rykotxet, 2025-01-15:
### Checks

- [X] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

sd2psx

### PS1 or PS2?

PS2

### [PS2 ONLY]: special consoles

None

### Console model

SCPH-90004

### Firmware version

1.0.0-beta12

### Explain your issue

I tried [DKWDRV](https://github.com/DKWDRV/DKWDRV) (loading from USB) to play PS1 games on PS2, and I had these issues with SD2PSX (beta12 firmware):
- Mode PS2 switches automatically to mode PS1 after launching a PS1 game, but the former PS2 game ID played on the console is still selected on SD2PSX. So a new folder for that PS2 game ID is created under PS1 parent folder, with a 128Kb MC channel file inside.
- No autodetection of game ID for the PS1 game running.

Note: "GameID" is "ON" under "[PS1]" tag in settings.ini (and "Autoboot" is "OFF").

bbsan2k, 2025-01-30:
Fixed

closed 2025-01-30 by bbsan2k

rykotxet, 2025-02-09:
Nice. I confirm the first point is solved.

The second point, auto Game ID in PS1, still not working. Is the issue in the DKWDRV side? I updated firm to 1.0.0, BTW.

commit 0a084cd3 (2025-01-30): Fix #11