sd2psXtd/firmware #12: [QUESTION]: IGR functionality

rykotxet, 2025-01-15:
### Checks

- [X] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

sd2psx

### PS1 or PS2?

PS2

### [PS2 ONLY]: special consoles

None

### Console model

SCPH-90004

### Firmware version

1.0.0-beta12

### Explain your issue

Could you explain further about that IGR functionality added in beta12?
If it's related to OPL's IGR, I tried and it didn't work to me. I've IGR path in OPL pointing to "mc:/APPS/OPNPS2LD.ELF", that it's inside my autoboot MC. After triggering IGR button combo in a PS2 game launched with OPL, MC didn't switch back to autoboot MC, so the IGR sent me to PS2 browser instead of back to OPL.

bbsan2k, 2025-01-29:
Right now, the reset functionality is only available in OPL MMCE beta 2. Once it is considered stable, the MMCE group will issue a PR to official OPL.

rykotxet, 2025-02-09:
@bbsan2k But how does it work? I tried it in OPL MMCE beta 2, and I couldn't figure out how to perform a proper IGR.

In the meanwhile, I found this [IGR app](https://ps2wiki.github.io/sas-apps-archive/sysappslistings/In-Game-Reset/In-Game-Reset.html) that loads autoboot on SD2PSX and reboots the PS2, but I need to place that IGR.ELF in a USB drive or another MC (in the other slot).