sd2psXtd/firmware #13: [REPORT]: Different outcomes when you turn on and reset your PS2

rykotxet, 2025-01-15:
### Checks

- [X] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

sd2psx

### PS1 or PS2?

PS2

### [PS2 ONLY]: special consoles

None

### Console model

SCPH-90004

### Firmware version

1.0.0-beta12

### Explain your issue

I don't know if it's intended or not, but I've to report it because it's a bit confusing for me. The thing is:
- When my PS2 is off, I push power button, PS2 turns on, and autoboot MC loads at start.
- When my PS2 is on, I push power button, PS2 resets, and the same MC that was loaded before (of a PS2 or PS1 game) is still selected instead of autoboot MC.

I my opinion, autoboot MC should be loaded also when you reset the PS2.

AKuHAK, 2025-01-15:
duplication of #5

bbsan2k, 2025-01-29:
Currently there is no way to detect a reset from sd2psx, unless it is sent by something like OPL MMCE edition.
This means, that pressing the reset button will never send you back to the boot card for the foreseeable future 😔

closed 2025-01-29 by bbsan2k