sd2psXtd/firmware #17: Doubt

Windows10User2, 2025-02-01:
If PSxMemCard users should install the sd2psx variant, then why was the PSxMemCard one released?

bbsan2k, 2025-02-01:
To clear things up:
- PSXMemCard : use the psxmemcard.uf2 or pmc+.uf2
- those variants don’t come with psram, so none is used
-  pmc zero is a flavor of pmc with different pinout

closed 2025-02-01 by bbsan2k

Windows10User2, 2025-02-01:
> To clear things up:
> 
>     * PSXMemCard : use the psxmemcard.uf2 or pmc+.uf2
> 
>     * those variants don’t come with psram, so none is used
> 
>     * pmc zero is a flavor of pmc with different pinout

[Here](https://github.com/sd2psXtd/firmware?tab=readme-ov-file#general-support-for-other-rp2040-based-mmce-devices) it says to use the PMC+ variant so could you update the text? So, is there a PicoMemcard+ Zero as well?