sd2psXtd/firmware #18: Doubts

Windows10User2, 2025-02-01:
If PSxMemCard users should install the sd2psx firmware variant, then why was the psxmemcard one released?

May PicoMemcard+ be based on RP2040-Zero as well (like PicoMemcard may be based on both RP2040 and RP2040-Zero)? If so, should the pmczero variant be installed in it (as well as on a RP2040-Zero based PicoMemcard) and the pmc+ variant on a RP2040 PicoMemcard+? And which variant should be installed on a RP2040 PicoMemcard?

rafer22, 2025-02-01:
use pmczero

Windows10User2, 2025-02-01:
> use pmczero

In which cases?

amak79, 2025-02-02:
I also noticed conflicting information regarding which firmware to use for the Bitfunx PSxMemCad (Gen 1). In the README under [General: Support for Other RP2040-Based MMCE Devices](https://github.com/sd2psXtd/firmware?tab=readme-ov-file#general-support-for-other-rp2040-based-mmce-devices) it says to use the PMC+ firmware, but the [Firmware Dowload](https://sd2psxtd.github.io/download.html) page links to `psxmemcard.uf2`.

Windows10User2, 2025-02-03:
> I also noticed conflicting information regarding which firmware to use for the Bitfunx PSxMemCad (Gen 1). In the README under [General: Support for Other RP2040-Based MMCE [...]

Do you know which firmware should be installed on the non-plus RP 2040 PicoMemcard RP and on the RP2040-Zero non-plus PicoMemcard? Is the non-plus PicoMemcard compatible with sd2psxTd?

amak79, 2025-02-04:
> Do you know which firmware should be installed on the non-plus RP 2040 PicoMemcard RP and on the RP2040-Zero non-plus PicoMemcard?

I believe it should be ` pmc+.uf2 ` for the RP2040 and `pmczero.uf2` for the RP2040-Zero. If you load the incorrect firmware it shouldn't "brick" the device. It just won't function correctly. In that case you can hold the 'BOOTSEL' or 'BOOT' button and plug your device into your computer and load the correct firmware. 

The issue I raised has been fixed in the develop branch in commit d5357072. The correct firmware for the Bitfunx PSxMemCad (Gen 1) is ` psxmemcard.uf2 `

Windows10User2, 2025-02-08:
> > Do you know which firmware should be installed on the non-plus RP 2040 PicoMemcard RP and on the RP2040-Zero non-plus PicoMemcard?
> 
> I believe it should be `pmc+.uf2` for the RP2040 and `pmczero.uf2` for the RP2040-Zero. If you load the incorrect firmware it shouldn't "brick" the device. It just won't function correctly. In that case you can hold the 'BOOTSEL' or 'BOOT' button and plug your device into your computer and load the correct firmware.
> 
> The issue I raised has been fixed in the develop branch in commit [d535707](https://github.com/sd2psXtd/firmware/commit/d5357072a8a9bb7aebb420d831dfc261ff6f0b40). The correct firmware for the Bitfunx PSxMemCad (Gen 1) is `psxmemcard.uf2`.

Now, they're saying [here](https://github.com/sd2psXtd/firmware?tab=readme-ov-file#general-support-for-other-rp2040-based-mmce-devices) to install the pmc+ variant again on PSxMemCard instead of the psxmemcard one.

bbsan2k, 2025-03-12:
PSXMemcard also uses the LED on those boards. Adding this for PMC+/PMCZero might be a problem, as the LED driver needs to be implemented first. It's on the Todo List, but low Priority atm.

Windows10User2, 2025-03-14:
> PSXMemcard also uses the LED on those boards. Adding this for PMC+/PMCZero might be a problem, as the LED driver needs to be implemented first. It's on the Todo List, but low Priority atm.

What LED?

bbsan2k, 2025-03-16:
Closed. Please read README.md

closed 2025-03-16 by bbsan2k

Windows10User2, 2025-03-24:
> Closed. Please read README.md

Already did and precisely due to that I made the questions I did.

Why did you hijack my issue to talk about a PSxMemCard LED when I didn't ask about it?