sd2psXtd/firmware #19: PicoMemcard PS1

Androxilogin, 2025-02-06:
I'm confused on the naming convention. Is it 'Card1-1.mcd', 'Card1-2.mcd' or 'Card1-1.mcd', 'Card2-1.mcd'? What are these numbers supposed to indicate? There's not a lot of documentation on this. Do the button combinations still work from the original project in order to switch PS1 cards back and forth and create new ones? (Start+Select+Up/Down/Triangle?)

svk77, 2025-03-03:
> Я запутался в правилах наименования. Это «Card1-1.mcd», «Card1-2.mcd» или «Card1-1.mcd», «Card2-1.mcd»? Что должны обозначать эти числа?

Card1-1.mcd ->  Card1 это название папки в которой находится этот образ карты памяти, -1 это порядковый номер карты памяти в этой папке.
Например, Game3-4.mcd - это образ карты памяти, который находится внутри папки Game3 и является там четвёртым по счёту.
На PicoMemcard с этим сложнее разобраться, потому что там нет дисплея. Я имею и PicoMemcard и PSxMemCard Gen2 (Bitfunx).

bbsan2k, 2025-03-17:
> Card1-1.mcd -> Card1 это название папки в которой находится этот образ карты памяти, -1 это порядковый номер карты памяти в этой папке. Например, Game3-4.mcd - это образ карты памяти, который [...]

Card1-1.mcd -> Card1 is the name of the folder in which this memory card image is located, -1 is the serial number of the memory card in this folder.

For example, Game3-4.mcd is a memory card image that is inside the Game3 folder and is the fourth in a row there.

It's harder to figure it out on PicoMemcard because there's no display there. I have both PicoMemcard and PSxMemCard Gen2 (Bitfunx).

-- Please use English here. Thanks for commenting :)

closed 2025-03-17 by bbsan2k

Androxilogin, 2025-03-20:
> > > Я запутался в правилах наименования. Это «Card1-1.mcd», «Card1-2.mcd» или «Card1-1.mcd», «Card2-1.mcd»? Что должны обозначать эти числа?
> > 
> > 
> > Card1-1.mcd -> Card1 это название папки в которой находится этот образ карты памяти, -1 это порядковый номер карты памяти в этой папке. Например, Game3-4.mcd - это образ карты памяти, который находится внутри папки Game3 и является там четвёртым по счёту. На PicoMemcard с этим сложнее разобраться, потому что там нет дисплея. Я имею и PicoMemcard и PSxMemCard Gen2 (Bitfunx).
> 
> Card1-1.mcd -> Card1 is the name of the folder in which this memory card image is located, -1 is the serial number of the memory card in this folder.
> 
> For example, Game3-4.mcd is a memory card image that is inside the Game3 folder and is the fourth in a row there.
> 
> It's harder to figure it out on PicoMemcard because there's no display there. I have both PicoMemcard and PSxMemCard Gen2 (Bitfunx).
> 
> -- Please use English here. Thanks for commenting 😊
> 
> Button Combos for card switching in >=1.1.1 are R1 + R2 + L1 + L2 + Arrow Keys Arrow keys are: Up: Card Up Down: Card Down Right: Channel Up Left: Channel DOwn

I'm still having trouble with the layout. With the original PicoMemCard, once it reached the end of the cards, it would not go any further; just stay on that card without reloading but it would blink to indicate that it recognized a command was sent. Is that the case here?

Another issue I'm having is when I add a card to the BOOT folder and define it with an ini, it always skips my FreePSXBoot image and goes right to the next card. I've tried having that as the only card in that folder and making a new folder with other cards and that still didn't work. Channels are folders, right?

Also, I haven't tested this, but within the memory card menu- if you accidentally delete a save game that you didn't want to, on a regular memory card you can hit the four top buttons rapidly after deleting and it will reappear. Does that function still work? This firmware is awesome, just confusing while using this device without having used one of the others at all.