sd2psXtd/firmware #22: [REPORT]: fr sd2psXtd in MODE = PS2 don't work on PS2 schp-90000

Lexxofff, 2025-03-01:
### Checks

- [x] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

sd2psx

### PS1 or PS2?

PS2

### [PS2 ONLY]: special consoles

None

### Console model

schp-90000

### Firmware version

1.0.0

### Explain your issue

I installed firmware sd2psXtd on PSxMemCardGen2.
I do everything according to the instructions: CIV.bin, etc.
My PS2 does not see the PS2 memory card when the PS2 mod is selected

usually firmware **sd2ps** PS2 see the PS2 memory card when the PS2 mod is selected
downloaded from here
sd2psx](https://github.com/bbsan2k/sd2psx_firmware or https://github.com/sd2psx/firmware/tags

My PS2 version information: 
Console:  schp-90000
Browser: 1.40
CD player: 2.00
PS Driver: 2.00
CD player: 3.11

ps.
send me a link on how to use debug firmware. I'll collect debug logs

### Debug Logs

```shell

```

bbsan2k, 2025-03-12:
Hi,
can you please try to re-install civ.bin and check if the issue persists?

bbsan2k, 2025-03-31:
Closed due to inactivity

Lexxofff, 2025-04-01:
I tried re-install civ.bin but that doesn't help
I bought a ps2 schp-77004 my PSxMemCardGen2 (fr 1.0.0) works in MODE = PS2. 
But  PSxMemCardGen2 (fr 1.0.0) don't work on the schp-90000. Usual memory card work good! 

how to use debug?
I have I have PSxMemCard Gen2 and PicoMemcard+(handmade)

Lexxofff, 2025-04-28:
I checked 1.2 release. fr didn't work with my 90000. PS2 90000 doesn't see psxmemcard gen2 
will maybe try debug...?

kenelm-ng, 2025-04-29:
I have the same issue with my SCPH-90000, PS2 doesn't see the psxmemcard gen2 at all.

bbsan2k, 2025-04-29:
Is this a modchipped 90k?
What about ps1 mode - is the card detected?
I‘m testing on a 90k usually (2.2 BIOS) and it is working fine for me…

kenelm-ng, 2025-04-29:
It is not modchipped 90k. 
The card was detected and working with PS1 games.
I think only some models of 90k is not working, just like MemCard Pro2.
This model is not the revision version of 90k model so no issue running FMCB and don't need open tuna.

Mine one is 
Browser: 1.40
CD Player: 2.00
PS Driver: 2.00
DVD Player: 3.11

Serial HJ0825172
3-282-432-01 P FOXC (bottom right)

bbsan2k, 2025-05-02:
We are currently working on this. It seems, this is affecting a very close serial range in 90k consoles.

Lexxofff, 2025-05-03:
Maybe will help, this is https://github.com/sd2psx/firmware/releases, MODE = PS2 work on my PS2 90000, it doesn't support no extended functionality.

kenelm-ng, 2025-05-04:
> Maybe will help, this is https://github.com/sd2psx/firmware/releases, MODE = PS2 work on my PS2 90000, it doesn't support no extended functionality.

This works for me as well! Thank you!
What do you mean by "it doesn't support no extended functionality"?
Can this change be merged into the sd2psXtd repository as well @bbsan2k ? Thank you.

bbsan2k, 2025-05-04:
sd2psx firmware is the original firmware, that sd2psXtd is based on. There has been a lot of refactoring since then, so figuring out what exactly causes this is like searching for a needle in a haystack :/

If it's not too much to ask:
Can you please check the release hre (https://github.com/bbsan2k/sd2psx_firmware/releases) if the issue is already there?

kenelm-ng, 2025-05-04:
Hi @bbsan2k, thanks for your explanation and patience. I’ve just tested the unofficial 0.9.3 release, and it still works in my 90K PS2—so it looks like the bug was introduced sometime in the past year. Hopefully, that helps narrow down the root cause?

Lexxofff, 2025-05-05:
unofficial 0.9.3 release works in my 90K PS2 too

xifize, 2025-05-05:
I also have an affected model and I was testing out different releases. v0.9.5-dev seems to work while 1.0.0-beta10 is not recognized.

bbsan2k, 2025-05-08:
Ooookay... To keep you updated:
We are still investigating this. I succesfully created a Frankenstein FW that should work for this special 90k serial number range. The problem with this, is that it most certainly won't work with MMCE, since timings are very different.
As this is also affecting MCP2, I'm working with Sickle to figure out what exactly is the root cause of that issue.

kenelm-ng, 2025-05-08:
Thanks @bbsan2k for the update — really appreciate you taking the time to investigate this issue. While it may still be unlikely to work with MMCE, it's great news that you've found a way to make it work and keep it up to date. I'm happy to help test any nightly or development builds if needed.

bbsan2k, 2025-05-25:
Can you please try latest nightly and give some feedback 😊 ? Thx

kenelm-ng, 2025-05-25:
Thanks so much, I will test it today and let you know how it goes!

bbsan2k, 2025-06-01:
@xifize  Could you please retry the latest nightly and report if MMCE has been fixed for you?