sd2psXtd/firmware #23: [REPORT]: psxmemcard.uf2 with tony hax international v.1.5.9

Playmakeree7, 2025-03-01:
### Checks

- [x] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

psxmemcard

### PS1 or PS2?

PS1

### [PS2 ONLY]: special consoles

None

### Console model

SCPH-9001

### Firmware version

1.1.0-RC

### Explain your issue

When using tonyhax international with psxmemcard.uf2, causes that the PSXMemCard is not detected by the console, therefore it does not make the slot change.

### Debug Logs

```shell

```

Playmakeree7, 2025-03-02:
Firmware 1.0.0 beta 12 freeze with tonyhax

bbsan2k, 2025-03-16:
Please try with 1.1.1/latest nightly and report back

Playmakeree7, 2025-03-29:
Now it works with the last nightly.

closed 2025-03-29 by Playmakeree7