sd2psXtd/firmware #26: PS1: Card Switch Controller Combo

Playmakeree7, 2025-03-03:
Does the PS1 Card Switch Controller Combo work on the psxmemcard gen 1?

Playmakeree7, 2025-03-03:
I just updated to version 1.1.0 and the button combination doesn't work, and it doesn't detect my saves either, I had to return to version 1.0.0 beta 12

Playmakeree7, 2025-03-12:
In versions 1.1.0, 1.1.1, and nightly, some games detect the memory card, while others don't, such as Spyro the Dragon.

In versions 1.1.0, 1.1.1, and nightly, the Card Switch Controller Combo doesn't work.

ramz33, 2025-03-13:
In version nightly i think my sdcards are not recognized, no diode light, only when powering console short blue blink. Can't copy saves from real mc to PSXmemcard, i got bitfunx edition. I'm check at model SCPH-1002 Playstation. Can't check combos coz no saves showing.

bbsan2k, 2025-03-17:
Please check with todays' nightly - should be fixed now!

Playmakeree7, 2025-03-17:
I just updated to the nightly version, and the Card Switch Controller Combo now works.

But it still doesn't detect saves for some games like Spyro the Dragon 1 using UniRom.

When using Tonyhax for some games that have APv2, it also doesn't detect the saves.

Playmakeree7, 2025-03-17:
By the way, when you change the card with the button combination, the memory card does not indicate anything with the LED light.

ramz33, 2025-03-17:
In my case witn newest nightly on PAL SCPH-1002 bios E (2.0) combos not working but i have PAL SCPH-7002 and combos are working with that model,  before was not working i forgot to write :). Also when changing cards with combos on MC/CDplayer screen it not indicate but after that when you chose MC option it blinks and enter MC manager.

Playmakeree7, 2025-03-29:
My gegege save was corrupted, now all that's left is for it to detect my spyro 1 save. Everything else works fine.

bbsan2k, 2025-04-23:
Fix by adding additional controller checks

closed 2025-04-23 by bbsan2k