sd2psXtd/firmware #29: [REPORT]: Can we please 🥺 enable the use of the pcb buttons on the Pico+ diy cards

psxdev1, 2025-03-07:
### Checks

- [x] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

pmc+

### PS1 or PS2?

PS2

### [PS2 ONLY]: special consoles

None

### Console model

Sch-any

### Firmware version

Beta 12

### Explain your issue

Please 🥺🥺🙏🥺

### Debug Logs

```shell

```

psxdev1, 2025-03-07:
[image]

bbsan2k, 2025-03-16:
Thats on my todo list for 1.2😊

bbsan2k, 2025-04-13:
I added this in commit ca21c6bbb727f5ce56bec9c7c11094392489919e - but could not get it to work on my devices.
Most certainly its wrong buttons. Can you please test and report back?

[firmware.zip](https://github.com/user-attachments/files/19724225/firmware.zip)

psxdev1, 2025-04-13:
It doesn't work for me 😭

bbsan2k, 2025-04-23:
fixed in latest nightly (84ffa26b9c3722fa57c3697d782a3155fe97070b)

closed 2025-04-23 by bbsan2k