sd2psXtd/firmware #32: [REPORT]: PS1 GameID broken on 1.1.1

ishioni, 2025-03-14:
### Checks

- [x] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

sd2psx

### PS1 or PS2?

PS1

### [PS2 ONLY]: special consoles

None

### Console model

SCPH-5502

### Firmware version

1.1.1

### Explain your issue

When running a game through xstation, i can see the GameID pop up for a second, but then the card switches to a generic one, and advances 2 cards (so if generic card 1 was set on boot, it will switch to card 3). I went back to 1.0.0 and can't replicate it there. PS2 works fine

### Debug Logs

```shell

```

bbsan2k, 2025-03-16:
Please try latest nightly and report if issue persists

ishioni, 2025-03-16:
I did last night and whilst the card didn't switch anymore, all games I tried (FF IX, Castlevania SOTN) reported the card was corrupted

bbsan2k, 2025-03-17:
This happens because of the images of the card on sd are corrupted. Please double check if they have the correct file size.
After removing them, Game ID on nighlty should work fine again

ishioni, 2025-03-17:
No, not really because downgrading to 1.0 and using the exact same file (the MMC wasn't created using nightly or 1.1.1) worked perfectly. But I'll try nonetheless. Does the debug version write the log to the SD card or some serial?

ishioni, 2025-03-18:
Okay i tried the newest nightly and it does appear that everything works fine now. I'll close this now

closed 2025-03-18 by ishioni