sd2psXtd/firmware #34: [REPORT]: SD2PSX starts making memory cards nonstop after game ID is received

Hansukie, 2025-03-24:
### Checks

- [x] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

sd2psx

### PS1 or PS2?

PS1

### [PS2 ONLY]: special consoles

None

### Console model

SCPH-1001 and 7501

### Firmware version

1.1.1 and latest nightly as of this post

### related game

Fatal Fury: Wild Ambition (USA)

### Explain your issue

This happens on v1.1.1 of sd2psx

When using Tonyhax International 1.5.9 to load a backup disc of Fatal Fury: Wild Ambition, after loading is finished and game ID is sent to the sd2psx, the sd2psx starts making new memory cards every second nonstop resulting in a runaway and can no longer be interacted with with unless power is cut. This happened on two separate PS1 consoles; 1001 and 7501. 

The latest nightly did not have the runaway issue but has a new issue; when save data for wild ambition is created and the console reset/powered off, the save data for FF: WA becomes corrupt. It still is recognizable by the game but says that the file is broken and attempts at loading the save results in "failure to load".  

### Debug Logs

```shell

```

bbsan2k, 2025-03-31:
Please re-test on latest nightly - issue should be fixed with this!

Hansukie, 2025-04-01:
Not sure what it is with the new nightly, but after installing it, my sd2psx locks up/freezes after a few seconds when it's connected and powered via the usb c port. 

I reinstalled v1.0.0 and it doesn't have the freezes anymore. Reinstalled the nightly to see if it was a fluke, same result. 

I didn't get a chance to test to see if the creating infinite memory cards bug was fixed.

bbsan2k, 2025-04-23:
Issue is finally fixed in latest nightly