sd2psXtd/firmware #35: [REPORT]: All virtual memory cards other than boot error out on ps1

ShutupAshley02, 2025-03-29:
### Checks

- [ ] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

psxmemcard

### PS1 or PS2?

PS1

### [PS2 ONLY]: special consoles

None

### Console model

SCPH-9001

### Firmware version

current nightly version as of 3/29/25

### related game

Castlevania Symphony of the night [SLUS-00067], Medal of Honor [SLUS-00974], Resident Evil Director's Cut DS Version [SLUS-00747], Resident Evil 2 DS version [SLUS-00748, SLUS-00756]

### Explain your issue

when flashed with current firmware the PSXMemcard Gen 1 can boot into FreePsXBoot just fine but when games are loaded they can either not view there is a memory card or the game displays that there's a memory card error, while in the file system of the sd card it is creating the correct virtual memory cards that are even accessible thought software

### Debug Logs

```shell

```

bbsan2k, 2025-03-30:
Confirmed!
Gonna look into that!

bbsan2k, 2025-03-31:
@ShutupAshley02  Please retest with latest nightly 😉 Should be fixed now!

commit 479018b2 (2025-03-31): Proper reset detection (fix #35)
closed 2025-04-02 by bbsan2k