sd2psXtd/firmware #38: [REPORT]: Failed to load PS1 save file on PS2 console.

unknown2918, 2025-04-05:
### Checks

- [x] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

sd2psx

### PS1 or PS2?

PS2

### [PS2 ONLY]: special consoles

None

### Console model

SCPH-30005R

### Firmware version

1.1.1 & nightly

### related game

Akumajou Dracula X: Gekka no Yasoukyoku [SLPM-87328]

### Explain your issue

When running the PS1 software Akumajou Dracula X on the PS2,
the save file cannot be loaded. An "error" message appears on the screen.
It loads normally in the memory card browser screen, but only happens when the game is running.

[image]

[image]

### Debug Logs

```shell

```

unknown2918, 2025-04-05:
I am using PSxMemCard Gen2

unknown2918, 2025-04-05:
This issue also occurs on SCPH 30005R and 50005 consoles.

Playmakeree7, 2025-04-05:
Same PSxMemCard Gen1, try firmware 1.0.0

unknown2918, 2025-04-06:
> Same PSxMemCard Gen1, try firmware 1.0.0

Does gen1 have the same issue?
It's the same even if you use 1.0.0

Playmakeree7, 2025-04-18:
Castlevania and Spyro still not working.

[image]

[image]

bbsan2k, 2025-04-27:
Issues should be resolved in 1.2.0

closed 2025-04-27 by bbsan2k