sd2psXtd/firmware #39: [REPORT]: Incompatibility with PS1 Multitap?

AkikoKumagara, 2025-04-11:
### Checks

- [ ] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

sd2psx

### PS1 or PS2?

PS1

### [PS2 ONLY]: special consoles

None

### Console model

SCPH-7501

### Firmware version

1.1.1

### Explain your issue

The card fails to correctly initialize when plugged into a PS1 Multitap (Official Sony in my case) rather than the console's own memory card port.

Normally, this wouldn't be a big deal, but I'm attempting to use my card along with a MiSTer FPGA as well as real hardware (PS1 and PS2 both) and need the Multitap to interpose the memory card to the FPGA system since it's the most convenient way for me to do so.

Is this incompatibility something that can be addressed via a firmware update, or is the hardware just functionally incompatible for some reason? See image attached for fatal error detail.

[image]

### Debug Logs

```shell

```

bbsan2k, 2025-04-17:
The problem with most SNACs on MiSter is, that they cannot provide enough current for sd2psx to work properly.

However: I talked to Taki, and we'll try to make sure, the SuperStation will be able to run the card properly - so once my preorder is fulfilled I can further dive into that and try to sort out the issues...

AkikoKumagara, 2025-04-17:
> The problem with most SNACs on MiSter is, that they cannot provide enough current for sd2psx to work properly.

Interesting, but my SCPH-7501 exhibits this behavior as well with the multitap.

bbsan2k, 2025-05-30:
Hmmm maybe the Multitap is limiting the current as well.
Unfortunately I don't own an original PS1 multitap...
You can try if latest firmware changes stuff, since there were some modifications on PS1 side, but since it's already failing to open the sd card, I don't have much hope for it.

AkikoKumagara, 2025-05-30:
> Hmmm maybe the Multitap is limiting the current as well. Unfortunately I don't own an original PS1 multitap... You can try if latest firmware changes stuff, since there were some modifications on PS1 side, but since it's already failing to open the sd card, I don't have much hope for it.

Thanks for the info. I've tried the latest nightly (at time of comment) now and here are my findings:

PsxMemCard Gen 2 + SCPH-50001 PlayStation 2 console + PlayStation 1 Multitap: WORKING
PsxMemCard Gen 2 + SCPH-7501 PlayStation 1 console + PlayStation 1 Multitap: NOT WORKING (same error)
PsxMemCard Gen 2 + MiSTer Pi FPGA (psx_shift240 core) + PlayStation 1 Multitap: NOT WORKING (same error)