sd2psXtd/firmware #42: [REPORT]: Latest 1.2.0-RC1 Doesn't Automatically Switch Memory Card When Loading from TonyHax

Hansukie, 2025-04-25:
### Checks

- [x] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

sd2psx

### PS1 or PS2?

PS1

### [PS2 ONLY]: special consoles

None

### Console model

SCPH-101

### Firmware version

1.2.0-RC1

### Explain your issue

Latest 1.2.0-RC1 Doesn't Automatically Switch Memory Card When Loading from TonyHax

Previous firmwares did automatically switch to the proper memory card after loading. 1.2.0-RC1 does not. It just stays on the boot card and you have to manually switch by pressing the buttons on the sd2psx. 

### Debug Logs

```shell

```

bbsan2k, 2025-04-27:
Can you please try the release version?

Hansukie, 2025-04-27:
> Can you please try the release version?

I tried with the release firmware 1.2.0 and it still doesn't work. When game ID is sent, the sd2psx stays on the bootcard instead of auto switching. The auto switching worked fine on 1.0.0 and 1.1.0.

bbsan2k, 2025-04-28:
Oh wait - you are referring to switching to the default card? This is not implemented, since if Game ID is disabled, the assumption is, that the card should NOT react to it!

bbsan2k, 2025-05-25:
Can you please check latest nightly?

bbsan2k, 2025-06-01:
Hey,
I checked again - in fact the behaviour on 1.1 and 1.0 was a bug, since game id was evaluated even if it shouldn't have been.
You can either use game id and add a custom mapping in Game2Folder or you need to manually switch - but switching to default card without game id being active is not the intended behaviour.

closed 2025-06-01 by bbsan2k