sd2psXtd/firmware #43: [REPORT]:

djdirtydeckx, 2025-04-26:
### Checks

- [ ] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

sd2psx

### PS1 or PS2?

PS2

### [PS2 ONLY]: special consoles

None

### Console model

scph-5004

### Firmware version

1.0.0

### related game

Gradius V [SLES_520.95]

### Explain your issue

Latest firmware 1.0.0. Gradius 5 won't load in any region. It's a black screen and won't start. Steel Dragon EX has issues with the Bluetooth controller. It does weird things and the D-pad doesn't respond properly. I'm using the official 256GB SanDisk in exFAT format. Greetings and thanks for this great firmware.

### Debug Logs

```shell

```

djdirtydeckx, 2025-04-26:
I have also tried firmware 1.1.1 with the same results in both games.

bbsan2k, 2025-04-29:
This is actually an OPL -MMCE issue - can you please open an issue in that repository?
Thanks!

closed 2025-04-29 by bbsan2k