sd2psXtd/firmware #44: [REPORT]: PS1 GameID not working - System with Blueretro and XStation

vsilvalopes, 2025-04-28:
### Checks

- [x] I have checked if a report of the same problem, in the same environment, is already listed [here](https://github.com/sd2psXtd/firmware/issues)

### firmware

sd2psx

### PS1 or PS2?

PS1

### [PS2 ONLY]: special consoles

None

### Console model

SCPH-5001

### Firmware version

1.2.0

### related game

None

### Explain your issue

Hi !
My system has an Xstation installed and an internal blueretro adapter, running the latest version.
The GameID feature on this version 1.2.0, is not working at all.

I reverted back to the "stock" Bitfunx firmware and the GameID feature is working just fine.
I can change a game reseting the console and on a coldbot, and the gameID works as expected.

I'm using this version of the Bitfunx firmware : https://github.com/gamebitfunx/PSxMemCardGen2/releases/tag/82bc0b3

Thanks in advance!

### Debug Logs

```shell

```

bbsan2k, 2025-05-25:
Hey,
I just added some fixes for game id stuff and PS1 card emulation in general. Can you please have a look if latest nightly does fix your issues?
Thanks 😊

inteyes, 2025-05-27:
I'm not the OP, but I want to inform that latest nightly solves this problem to me. Thank you!

bbsan2k, 2025-05-30:
Nice! Thanks for testing!